ros2/demos #464: inquiry about implementation of ${${PROJECT_NAME}_EXPORTED_TARGETS} in ROS2

rjshim, 2020-09-16:
Hi, 

## Feature request
#### Feature description
As the title says, I wonder if ${${PROJECT_NAME}_EXPORTED_TARGETS} in ROS1 has also been implemented in ROS2 🤔. I am trying to generate a message/service and use it in the same package but I cannot find any sample demo codes doing it.

Ryan

clalancette, 2020-09-16:
There is an example that does it here: https://github.com/ros2/demos/tree/master/logging_demo .  In particular, the CMake function that you need to call is at this line: https://github.com/ros2/demos/blob/2b94e70a06d9a0b37ba6b7d3405e5e2d1779e509/logging_demo/CMakeLists.txt#L34 .

Note that we typically don't recommend putting the message/service file in the same package as the code.  If they are in the same package, then an external node that wants to contact your node needs to install the entire node package, not just a message package.  This is why you don't see many examples of it in the ROS 2 codebase.

I'm going to go ahead and close this, assuming this answered the question.  In the future, I'll suggest you open questions like this on https://answers.ros.org, which is a central resource for searchable questions and answers.

closed 2020-09-16 by clalancette

rjshim, 2020-09-17:
@clalancette Thanks a lot! I will.  Ryan