ros2/demos #483: sym

JaehyunShim, 2021-01-27:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- Installation type:
  -  source
- Version or commit hash:
  -  latest master branch
- DDS implementation:
  -  FastRTPS
- Client library (if applicable):
  -  rclpy

#### Steps to reproduce issue

1. add a launch file as this [link](https://github.com/ros2/demos/blob/master/topic_monitor/setup.py#L17)
2. colcon build --symlink-install

#### Expected behavior
A symbolic link should be added to /path_to_install_directory/package_name/share/package_name/launch/

#### Actual behavior
1. Not a symbolic link but a copy of the launch file is added.
2. The file is not added to /path_to_install_directory/package_name/share/package_name/launch/, but added to /path_to_install_directory/package_name/share/package_name/

if you add param or config files, the path to the files will be more important so I would like to ask if there is any way to add the entire param/config/launch directories using rclpy. 

[image]

clalancette, 2021-01-30:
If I remember properly, this is a limitation of the python `setup.py` installation.  That is, I seem to remember that there is no way to convince `setup.py` to install with symlinks.

It might be possible to do something about this (by either enhancing setup.py upstream, or doing some work in colcon).  For that reason, I think this issue should be closed and moved over to colcon (probably https://github.com/colcon/colcon-python-setup-py).

JaehyunShim, 2021-02-01:
@clalancette  

Thank you very much for your reply. What about the second point, i.e., adding directories including launch/param/etc files to the shared directory, not adding each of the files? 

Thank you very much in advance, 
Jaehyun

clalancette, 2021-02-01:
> Thank you very much for your reply. What about the second point? ( i.e., adding directories including launch/param/etc files to the shared directory, not adding each of the files)

I would suspect that is possible with `setup.py`, but I'd have to play around with it.

clalancette, 2021-02-18:
I'm actually going to close this one in favor of the one you opened against colcon-python-setup-py, which is the appropriate place.

closed 2021-02-18 by clalancette

JaehyunShim, 2021-02-19:
Did anything work for this? 

> I would suspect that is possible with `setup.py`, but I'd have to play around with it.

Thanks,
Jaehyun