ros2/demos #503: [topic_monitor] Uses 100% of a core

mjeronimo, 2021-04-20:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: Debian-based install
- DDS implementation: any

#### Steps to reproduce issue
* In one terminal: ros2 run topic_monitor topic_monitor 
* In another terminal: top

#### Expected behavior
The topic_monitor uses a reasonable amount of CPU. 

#### Actual behavior
Running the topic_monitor uses 100% of a CPU.

clalancette, 2021-04-21:
It's probably worthwhile to see if https://github.com/ros2/demos/pull/461 fixes or improves the situation here.

mjeronimo, 2021-04-22:
Yes, it does. I pulled the branch from #461, rebased against the latest code (C++17 changes required) and ran the following:

`ros2 launch topic_monitor fragmentation_demo.launch.py`

The CPU utilization was about 10% on my system with the fix versus about 100% without.

clalancette, 2021-04-22:
OK, great.  I'm going to close this as a duplicate of #460 then.

closed 2021-04-22 by clalancette