ros2/demos #519: std_msgs_conversion/int32_t.hpp: No such file or directory

fujitatomoya, 2021-05-17:
## Bug report

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Source
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/3d98527067e4915933a99b4b7f89a4fbe3039592
- DDS implementation:
  - N.A
- Client library (if applicable):
  - N.A

#### Steps to reproduce issue

- https://docs.ros.org/en/rolling/Installation/Ubuntu-Development-Setup.html

```bash
# colcon build --symlink-install
...<snip>
--- stderr: intra_process_demo
/root/ros2_ws/colcon_ws/src/ros2/demos/intra_process_demo/src/two_node_pipeline/two_node_pipeline.cpp:23:10: fatal error: std_msgs_conversion/int32_t.hpp: No such file or directory
   23 | #include "std_msgs_conversion/int32_t.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

#### Expected behavior

build successfully.

#### Actual behavior

failed to build.

#### Additional information

related to https://github.com/ros2/demos/commit/29f74d798afc347bf6efddcfef8722dae626184b

fujitatomoya, 2021-05-17:
@audrow friendly ping. i tried to do full build a couple of time with different machines, but so far i am unable to complete full build.

clalancette, 2021-05-17:
I think this will be solved with #520.

fujitatomoya, 2021-05-18:
@clalancette confirmed it's been fixed.

closed 2021-05-18 by fujitatomoya