ros2/demos #531: Add README

lorepieri8, 2021-08-10:
It would be good to have a README file for each demo that briefly explains what's inside.

audrow, 2021-12-20:
I think this is a good idea.

cardboardcode, 2022-07-01:
+1 from me as well. I should be available to take a look at this within the month of July 2022 if no one else is doing it. 

@lorepieri8 @audrow 

Will update accordingly here in this thread regarding progress.

cardboardcode, 2022-07-01:
## TODO
For each demo to have its own `README.md`, a total of **18** `README.md` will need to be created for the following folders.

Updated as of `1st July 2022`:

- [ ] root `README.md`
- [ ] [action_tutorials](https://github.com/ros2/demos/tree/rolling/action_tutorials)/action_tutorials_cpp
- [ ] [action_tutorials](https://github.com/ros2/demos/tree/rolling/action_tutorials)/action_tutorials_interfaces/
- [ ] [action_tutorials](https://github.com/ros2/demos/tree/rolling/action_tutorials)/action_tutorials_py
- [ ] [demos](https://github.com/ros2/demos)/composition/
- [ ] [demos](https://github.com/ros2/demos)/demo_nodes_cpp/
- [ ] [demos](https://github.com/ros2/demos)/demo_nodes_cpp_native/
- [ ] [demos](https://github.com/ros2/demos)/demo_nodes_py/
- [ ] [dummy_robot](https://github.com/ros2/demos/tree/rolling/dummy_robot)/dummy_map_server/
- [ ] [dummy_robot](https://github.com/ros2/demos/tree/rolling/dummy_robot)/dummy_robot_bringup/
- [ ] [dummy_robot](https://github.com/ros2/demos/tree/rolling/dummy_robot)/dummy_sensors/
- [ ] [dummy_robot](https://github.com/ros2/demos/tree/rolling/dummy_robot)/dummy_sensors/
- [ ] [demos](https://github.com/ros2/demos)/image_tools/
- [ ] [demos](https://github.com/ros2/demos)/intra_process_demo/
- [x] https://github.com/ros2/demos/tree/rolling/lifecycle
- [ ] [demos](https://github.com/ros2/demos)/lifecycle_py/
- [ ] [demos](https://github.com/ros2/demos)/logging_demo/
- [ ] [demos](https://github.com/ros2/demos)/pendulum_control/
- [ ] [demos](https://github.com/ros2/demos)/pendulum_msgs/
- [x] https://github.com/ros2/demos/tree/rolling/quality_of_service_demo
- [x]  https://github.com/ros2/demos/tree/rolling/topic_monitor
- [x] https://github.com/ros2/demos/tree/rolling/topic_statistics_demo

kagibson, 2022-07-21:
@cardboardcode What is the status with this? I would be happy to work on this or help

cardboardcode, 2022-07-21:
> @cardboardcode What is the status with this? I would be happy to work on this or help

Hi @kagibson, I have yet to get started. If you like, you can choose from the above list the tasks you wish to work on first. I would simply do the remaining. 

Thanks for the offer of help :blush:

cardboardcode, 2022-07-22:
@kagibson 

For now, I will get started doing the following tasks:

- [dummy_robot/dummy_map_server/](https://github.com/ros2/demos/tree/rolling/dummy_robot/dummy_map_server)
- [dummy_robot/dummy_robot_bringup/](https://github.com/ros2/demos/tree/rolling/dummy_robot/dummy_robot_bringup)
- [dummy_robot/dummy_sensors/](https://github.com/ros2/demos/tree/rolling/dummy_robot/dummy_sensors)
- [demos/image_tools/](https://github.com/ros2/demos/tree/rolling/image_tools)

Will update in the task list once it is completed. Aiming to get it done by **22nd July 2022**.

cardboardcode, 2022-07-22:
Hi @audrow, may I get your assistance to **review** the newly added and verified [image_tools - README.md](https://github.com/cardboardcode/demos/tree/dev/image_tools)?

Thank you.

kagibson, 2022-07-22:
@cardboardcode Thank you! I will work on the following:
- action_tutorials/action_tutorials_cpp/
- action_tutorials/action_tutorials_interfaces/
- action_tutorials/action_tutorials_py

kagibson, 2022-07-23:
@audrow I'm in the process of writing the readme for the action_tutorials. So far I have an introduction on what is going on in the example. Do we want an in depth description of the code (what each callback is doing etc) or just an explanation of what the demo is doing and how to run it?

cardboardcode, 2022-07-23:
@kagibson 

Awaiting feedback from @audrow on aforementioned completed `README.md`s. 

Will be starting work on the following new tasks:
- [demos/intra_process_demo](https://github.com/ros2/demos/tree/rolling/intra_process_demo)
- [demos/pendulum_msgs](https://github.com/ros2/demos/tree/rolling/pendulum_msgs)
- [demos/logging_demo](https://github.com/ros2/demos/tree/rolling/logging_demo)

audrow, 2022-07-25:
> @audrow I'm in the process of writing the readme for the action_tutorials. So far I have an introduction on what is going on in the example. Do we want an in depth description of the code (what each callback is doing etc) or just an explanation of what the demo is doing and how to run it?

@kagibson, I would think that a high level description would be enough, maybe referring to blocks of code and explaining what they're doing. That being said, if you find anything surprising or unintuitive, that's probably a good spot to go into more detail.

audrow, 2022-07-25:
> Hi @audrow, may I get your assistance to **review** the following added `README.md`s ?
> 
> * [image_tools - README.md](https://github.com/cardboardcode/demos/tree/dev/image_tools)
> * [dummy_map_server - README.md](https://github.com/cardboardcode/demos/tree/dev/dummy_robot/dummy_map_server)
> * [dummy_sensors - README.md](https://github.com/cardboardcode/demos/tree/dev/dummy_robot/dummy_sensors)
> * [dummy_robot_bringup - README.md](https://github.com/cardboardcode/demos/tree/dev/dummy_robot/dummy_robot_bringup)
> 
> Thank you.

@cardboardcode, thanks for doing those! Would you make a PR for those changes? It'll be an easier way to give you feedback. Probably the easiest way will be to make a PR for each README, since it will be easier to get them in one at a time as they're ready, rather than waiting on all of them to look good all at once.

cardboardcode, 2022-07-26:
@audrow. Noted. Will close PR (#531) and spilt the newly added `README.md`s into their own individual Pull Requests. 

Will update in this comment once that is done.

kagibson, 2022-07-26:
@audrow I've created a PR (#576) for the action_tutorials README's. Thank you!

SrijaneeBiswas, 2023-02-27:
@lorepieri8 @audrow Any more help wanted on this?

Would love to contribute as a first issue.

audrow, 2023-03-27:
Hi @SrijaneeBiswas, thanks for asking and sorry for the slow reply. I'm not aware of much in this repo that needs improvements, @cardboardcode  has done a great job.

However, there are plenty of things that could use improvement in the ros2_documentation repo. Here's a list of issues with help wanted. If you want to try any of them, feel free to @ me as a reviewer!
https://github.com/ros2/ros2_documentation/issues?q=is%3Aopen+is%3Aissue+label%3A%22help+wanted%22