ros2/demos #533: Lifecycle demo is broken

christophebedard, 2021-09-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 Focal
- Installation type:
  - Rolling from source
- Version or commit hash:
  - `master` (from https://github.com/ros2/ros2/blob/master/ros2.repos)
- DDS implementation:
  - Cyclone DDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Just follow the demo instructions: https://github.com/ros2/demos/tree/master/lifecycle#the-demo

```
# Terminal 1
$ ros2 run lifecycle lifecycle_talker
# Terminal 2
$ ros2 run lifecycle lifecycle_listener
# Terminal 3
$ ros2 run lifecycle lifecycle_service_client
```

#### Expected behavior

As described below the demo instructions, the client goes through a list of transitions and activates + deactivates the talker a few times.

#### Actual behavior

The client triggers transition 1 (to configure the talker), but it exits right after that.

It builds fine of course.

#### Additional information

Most likely unrelated to #504

I've had a lifecycle-related test start failing around August 23rd/24th in `ros2_tracing`: https://gitlab.com/ros-tracing/ros2_tracing/-/issues/124. It uses the same util functions as the client in the demo here to run through a list of transitions (https://gitlab.com/ros-tracing/ros2_tracing/-/blob/master/test_tracetools/src/test_lifecycle_client.cpp), which is why I checked if this demo worked.

That lifecycle test and the demo here both work with Galactic from source (https://github.com/ros2/ros2/blob/galactic/ros2.repos). It might be from the boilerplate/util code itself, but it hasn't changed in a while. Timewise, the failures seem to coincide with https://github.com/ros2/rclcpp/pull/1734. Reverting it (https://github.com/ros2/rclcpp/commit/679fb2ba334971d9769b44258df9095025567559) fixes the demo here and the lifecycle test in `ros2_tracing`.

christophebedard, 2021-09-30:
@ivanpauno since you made that change (https://github.com/ros2/rclcpp/pull/1734), do you have any clues or tips to provide? Otherwise I'll start the real debugging :laughing:

ivanpauno, 2021-10-01:
I would bet the issue is the demo is calling `future.get()` many times.
Now we're returning a `std::future` instead of a `std::shared_future`, the first one consumes the inner value, the second one returns a reference.

I guess the demo doesn't have a test.

---

I thought I already added that to the release notes, but it doesn't seem to be the case.
I will add that.

christophebedard, 2021-10-01:
Ah I see! Turns out to be a rather simple fix IIUC. I'll open a PR. Thank you!

PR #534 (2021-10-03): Fix use of future in lifecycle demo
Fixes #533

See https://github.com/ros2/demos/issues/533#issuecomment-932228044
closed 2021-10-03 by clalancette