ros2/demos #553: Bad order of node.destroy_node()  and rclpy.try_shutdown() in demos/demo_nodes_py/demo_nodes_py/topics/listener.py

hodnajit, 2022-01-21:
I found out, that in this example, there is bad order in

```
node.destroy_node() 
and rclpy.try_shutdown()
```

in demos/demo_nodes_py/demo_nodes_py/topics/listener.py 

I think, it should be fixed :)

clalancette, 2022-01-21:
Looking at the code in https://github.com/ros2/demos/blob/master/demo_nodes_py/demo_nodes_py/topics/listener.py, it looks like we do:

1. init
2. create node
3. use node
4. destroy node
5. shutdown

That seems to be the correct order to me.  Can you explain more about why you think it is the incorrect order?

hodnajit, 2022-01-21:
Thanks for answering that fast :) 
I am confused, that in https://github.com/ros2/demos/blob/master/demo_nodes_py/demo_nodes_py/topics/talker.py

The order of destroying node and shutting rcl is different
```
rclpy.try_shutdown()
node.destroy_node()
```

What is the correct way?

clalancette, 2022-01-21:
Ah, good catch.  In general, it is best to destroy resources in the reverse order that you allocated them.  In this case, the talker is wrong.  Would you mind submitting a pull request reversing the order in the talker?

clalancette, 2022-01-24:
#555 should fix this.

closed 2022-01-24 by clalancette

hodnajit, 2022-01-24:
Thank you very much :) Great work

PR #555 (2022-01-24): Cleanups in demo_nodes_py.
The following fixes are made:

* Ensure that the node is destroyed before calling rclpy.shutdown()
* Make sure that shutdown always uses rclpy.try_shutdown()
* Make sure to catch KeyboardInterrupt errors everywhere

Fixes #553