ros2/demos #560: Launch composable nodes (how-to): The program 'burgerimage' received an XWindow System error.

rfzeg, 2022-05-03:
## Bug report

I'm following the tutorial at https://docs.ros.org/en/galactic/How-To-Guides/Launching-composable-nodes.html. When I run the launch file I get the following error message:

```
[component_container-1] (burgerimage:22832): Gdk-ERROR **: 15:24:40.279: The program 'burgerimage' received an XWindow System error.
[component_container-1] This probably reflects a bug in the program.
[component_container-1] The error was 'BadAccess (attempt to access private resource denied)'.
[component_container-1]   (Details: serial 255 error_code 10 request_code 130 (MIT-SHM) minor_code 1)
[component_container-1]   (Note to programmers: normally, X errors are reported asynchronously;
[component_container-1]    that is, you will receive the error a while after causing it.
[component_container-1]    To debug your program, run it with the GDK_SYNCHRONIZE environment
[component_container-1]    variable to change this behavior. You can then get a meaningful
[component_container-1]    backtrace from your debugger if you break on the gdk_x_error() function.)
[ERROR] [component_container-1]: process has died [pid 22832, exit code -5, cmd '/opt/ros/galactic/lib/rclcpp_components/component_container --ros-args -r __node:=image_container -r __ns:=/'].
```

**Required Info:**

- Operating System:
  Ubuntu 20.04.4 LTS
- Installation type:
  From source `git clone https://github.com/ros2/demos.git -b galactic`   
- Version or commit hash:
  Output of git rev-parse HEAD `ed6a3eb69c78e01fe0133a5265c3ff252ac9a3cb`
- DDS implementation:
  The default rmw implementation
- Client library (if applicable):
  rclpy

#### Steps to reproduce issue

1)Create a new ROS 2 package, for instance:  
`ros2 pkg create --build-type ament_python composable_nodes_launcher --dependencies rclpy image_tools`

2) Create a new launch file and paste the code from the tutorial onto that file:  

```
import launch
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    """Generate launch description with multiple components."""
    container = ComposableNodeContainer(
            name='image_container',
            namespace='',
            package='rclcpp_components',
            executable='component_container',
            composable_node_descriptions=[
                ComposableNode(
                    package='image_tools',
                    plugin='image_tools::Cam2Image',
                    name='cam2image',
                    remappings=[('/image', '/burgerimage')],
                    parameters=[{'width': 320, 'height': 240, 'burger_mode': True, 'history': 'keep_last'}],
                    extra_arguments=[{'use_intra_process_comms': True}]),
                ComposableNode(
                    package='image_tools',
                    plugin='image_tools::ShowImage',
                    name='showimage',
                    remappings=[('/image', '/burgerimage')],
                    parameters=[{'history': 'keep_last'}],
                    extra_arguments=[{'use_intra_process_comms': True}])
            ],
            output='both',
    )

    return launch.LaunchDescription([container])
```

3) Open the `setup.py` file and add the imports and the line `(os.path.join('share', package_name, 'launch'), glob('launch/*.launch.py')),` within the data_files brackets.  

4) Compile, source and launch.

#### Expected behavior

The example doesn't crash.

#### Actual behavior

The example crashes with the error message shown above.

#### Additional information

clalancette, 2022-05-19:
This is likely a misconfiguration of the X-session somehow; that's how that `BadAccess` usually happens.  I unfortunately don't have a lot of good ideas on how to solve it, but make sure that you can run other windowed programs from the same prompt before trying that launch file again.

I don't know why you have to ignore `lifecycle`; that shouldn't be required either.

jasonbeach, 2022-08-02:
I have seen the issue a number of times with OpenCV GUI components.  For me it's always been inconsistent. one run I get this error, I run again all is fine. Are you by chance running ros2 in a docker container? I've heard adding `--ipc=host` to the docker run command may fix the issue, but I haven't had a chance to try it yet.