ros2/demos #612: RGB order flipped when using cam2image with showimage

jjd9, 2023-05-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 (Jammy)
- Installation type:
  - Debian apt install (binaries?)
- Version or commit hash:
  - ros-iron-image-tools/jammy,now 0.27.0-2jammy.20230429.031022 amd64 
- DDS implementation:
  - tested with Fast-RTPS and RTI Connext
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Open two terminals and source the ros2 space in each (i.e. `source /opt/ros/iron/setup.bash`) 

In terminal 1, run the cam2image executable
```
ros2 run image_tools cam2image --ros-args -p show_camera:=true
```

In terminal 2, run the showimage executable
```
ros2 run image_tools showimage
```

#### Expected behavior
The image being published and displayed by cam2image matches the image displayed by showimage.

#### Actual behavior
cam2image reads a BGR image from the camera, publishes it as a CV_U8C3, cv::Mat image, which does not preserve the channel ordering information. 
showimage receives the image and assumes that since it is a CV_U8C3, cv::Mat, it should convert it from RGB to BGR. But since that is not true, the result is an image whose channel ordering is flipped.

Here is the output I am getting (Left image: cam2image visual, Right image: showimage visual)

[image]

Thank you for your time! (-:

Yadunund, 2023-05-09:
I encounter this problem as well. Perhaps the solution is to add a parameter to let users override the encoding in `cam2image`.  Somewhat similar to https://github.com/ros-perception/vision_opencv/pull/505

EwingKang, 2024-03-04:
I encountered a similar problem with the `ros2 run image_tools showimage` tool. The source image is a ROS image converted from `cv_bridge::CvImage`, with proper "bgr8" desgination. The RVIZ can display the image correctly while the `showimage` tool shows reversed R-B channel.

sndsh404, 2026-11-11:
Hi, I tried running `cam2image` and `showimage` from the `osrf/ros:rolling-desktop` Docker image using my ThinkPad's built-in webcam.

I noticed that the colors looked wrong in the `showimage` window. Red and blue were swapped, so the skin looked blue.

I looked into `showimage.cpp` (lines 191-192) and found that it assumes every 3-channel image is `rgb8` and swaps the colors. But `cam2image` is already publishing images as `bgr8`. I tried commenting out the conversion, and that fixed the colors for my webcam, but I think that would break real `rgb8` images.

I think the issue is that `ROSCvMatContainer` loses the original encoding when converting the image to `cv::Mat`. My idea is to preserve the encoding so `showimage` knows whether it needs to swap the colors or not.

I also saw the suggestion in the original issue about adding a parameter to override the encoding. Which approach do you think would be better? Would it be okay if I worked on this?