ros2/demos #625: Do not use "off" for client_configure_introspection param in service introspection client

Yadunund, 2023-05-11:
As detailed here https://github.com/osrf/ros2_test_cases/issues/885#issuecomment-1539212847

clalancette, 2023-05-11:
I have to disagree with this one.  The underlying data structures use the terminology `off`, `metadata`, `content`, so I think this demo should use the same thing.

This issue isn't specific to this demo; basically any value that YAML considers to be "false" will be considered a boolean.

As @jacobperron pointed out, there are actually a couple of different workarounds for this.  One is to quote it:

```
ros2 param set /introspection_client client_configure_introspection \"off\"
```

Another option is to use the explicit string configuration from YAML:

```
ros2 param set /introspection_client client_configure_introspection '!!str off'
```

Maybe we should just make this explicit in the test case instead?

jacobperron, 2023-05-11:
@clalancette I don't recall a discussion about using transient local durability. The proposed design document suggest we should use the default QoS for service event topics (https://github.com/ros-infrastructure/rep/pull/360):

> The service event topics proposed in this REP shall use the default quality of service settings.

jacobperron, 2023-05-11:
Yeah, it would be nice to use the same terminology. I'm torn, since it would be nice to avoid this quirk of the demo. I suppose it's not a big deal since it's just a demo.

If someone wanted to make this feature configurable during runtime (like the demo) in their own application (or as part of the core ROS implementation), what could we suggest?  It seems like "off" might be a poor name choice given how integrated YAML is in ROS.

PR #634 (2023-06-07): Change the service introspection parameter off value to 'disabled'
With this we can avoid the tricky bits around YAML interpretation of 'off' as a boolean.

This should fix #625
closed 2023-06-07 by clalancette