ros2/demos #636: Add py versions of nodes from logging_demo pkg

Yadunund, 2023-05-23:
Presently there are some CPP based nodes in the logging_demo pkg while py versions do not exist.
Also, we recently added newer logging service related nodes in demo_nodes_cpp and demo_nodes_py. We could consider deprecating the logging_demo pkg and move the CPP nodes from there into demo_nodes_cpp. At the same time port py versions of these nodes into demo_nodes_py.

yyashikasharmaa, 2026-09-18:
Hi! I’d like to work on this. I have hands-on ROS 2 experience and would like to start with a small, reviewable Python port rather than mixing the package-deprecation work into the same PR.

My proposed first step is to port the behavior of logging_demo/src/logger_usage_component.cpp into demo_nodes_py/demo_nodes_py/logging/, add the corresponding console entry point, and add/adjust tests so the Python demo is exercised. After that, the logger-config example / C++ relocation can be handled as a separate PR if maintainers still want that direction.

Would that scoped first PR match what you want for this issue?

fujitatomoya, 2026-09-23:
@Yadunund can you confirm what/how needs to be changed above? sounds good to me, but wanted to confirm cz this is a few years old issue before implementation.

Yadunund, 2026-09-28:
@yyashikasharmaa @fujitatomoya addressing this ticket would still be valuable.

Some things to keep in mind
1. Let's move `LoggerUsage` and drop [LoggerConfig](https://github.com/ros2/demos/blob/810046ac184978147d38db39f20b882d0cc1b583/logging_demo/include/logging_demo/logger_config_component.hpp#L30) since the later is redundant with `enable_logger_service` + `rcl_interfaces/srv/SetLoggerLevels`
2. A few tutorials in https://github.com/ros2/ros2_documentation reference `logging_demo_main`. We'll need to target a PR there to update.
3. We can deprecate `logging_demo` for M turtle and remove it after. This way anyone referencing `logging_demo/srv/ConfigLogger` would be notified.