ros2/demos #642: When does CycloneDDS trigger message lost event?

mhaukeland, 2023-06-19:
In the quality of service [demo ](https://github.com/ros2/demos/tree/rolling/quality_of_service_demo#message-lost-status-event-demo) it is stated that: "CycloneDDS partially implements the feature (message lost event) and it only triggers the event under some limited circumstances"  Can anyone answer when CycloneDDS actually triggers the message lost event? And maybe even compare it to the FASTRTPS_DDS or the ConnextDDS?

clalancette, 2023-06-29:
@eboasson Could you take a quick look and see if this is a question you could answer?  Thanks.

eboasson, 2023-06-29:
@clalancette happy to.

@mhaukeland I can't compare it to FastDDS or Connext, for the simple reason that I don't know those implementations well enough. What I do know is that Cyclone DDS currently does not report sample lost events, and that this is a known issue.

Note that the issue is not *detecting* sample lost events, as deciding to skip sequence numbers is quite fundamental to the protocol stack. So the issue is really just setting the status/invoking the listener when it happens.

I have a number of reasons for wanting to fix it soon, but I have to admit that I sometimes stretch the concept of "soon" to the breaking point ... Please ping me if you don't hear something about it soon so I don't forget 🙂