ros2/demos #662: Exception in get_state: std::future_error: No associated state

AaronLPS, 2023-12-18:
https://github.com/ros2/demos/blob/69aa0e4d8662b7bd06138462cc328415b3198263/lifecycle/src/lifecycle_service_client.cpp#L124C25-L124C38

The std::future::get() function can only be called once as it will invalidate the std::future after it's called. In this code, future_result.get() is called multiple times.

fujitatomoya, 2023-12-19:
that is why we are using `std::shared_future` here?

https://github.com/ros2/demos/blob/69aa0e4d8662b7bd06138462cc328415b3198263/lifecycle/src/lifecycle_service_client.cpp#L108

or are you receiving the `std::future_error` exception with this demo in your environment?

AaronLPS, 2023-12-19:
`std::shared_future` solves the issue. I was looking at the outdated demo version in Foxy branch, which uses `auto future_result = client_get_state->async_send_request(request);`

closed 2023-12-19 by AaronLPS