ros2/demos #700: Single threaded executor is used for callbacks of two nodes

sayali-purdue, 2024-07-25:
## Bug report

**Required Info:**

- Operating System:
  - N/A
- Installation type:
  - Static analysis of source code
- Version or commit hash:
  - Latest
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Static review/inspection of source code
```

```

#### Expected behavior

#### Actual behavior

#### Additional information
In file [introspection.py](demo_nodes_py/demo_nodes_py/services/introspection.py), single-threaded executor is used to process callbacks of two ROS nodes. That means single thread is responsible for executing callbacks from two different callback groups (one default callback group per node). As discussed in ROS documentation, callbacks from different callback groups can execute in parallel. However due to single thread, the callbacks of the two nodes can not execute in parallel. Using a multi-threaded executor would be more efficient for this use case. 

I would appreciate to hear your thoughts on this.

fujitatomoya, 2024-07-26:
@sayali-purdue thanks for posting issue.

> In file [introspection.py](https://github.com/ros2/demos/issues/demo_nodes_py/demo_nodes_py/services/introspection.py)

this is 404, i think you meant https://github.com/ros2/demos/blob/rolling/demo_nodes_py/demo_nodes_py/services/introspection.py

> However due to single thread, the callbacks of the two nodes can not execute in parallel. Using a multi-threaded executor would be more efficient for this use case.

your assumption is correct, `SingleThreadedExecutor` cannot process the data in concurrent at the same time.

but with this example, i think timer, service request/response processes are sequentially happening.
besides, this is just an example to show the service introspection, so not time sensitive application to process the event as fast as possible.

there will be no problem to use `MultiThreadedExecutor` here, but i would think that is not just necessary here as an example.

clalancette, 2024-07-26:
Yes, agreed with @fujitatomoya .  The point of this example is not to show how to do things in parallel, but instead to show how service introspection works.  Thus, there is no need to complicate this example with more callback groups and a multi-threaded executor.

closed 2024-07-26 by clalancette